bluenviron/goroslib #3: ROS Topic incorrect type published?

jessedusty, 2020-03-27:
Hi aler9,

I was planning on using your library in my [senior design project](http://sad-bird.com), but I came across a bug. 

I tried publishing a topic from one of the examples (examples/publisher.go) and subscribing to it from `rostopic echo` and it gave me the following error message. 

```
ERROR: Cannot load message class for [goroslib/Msg]. Are your messages built?
```

`rostopic list -v` also shows the type of the message being `[goroslib/Msg]` even though the node should be publishing a `sensor_msg/Imu`. 

This can be reproduced using the following docker commands

Start the ros core:
```
docker run --rm -it --network=host --name=ros ros:melodic-ros-core-bionic roscore
```

Compile and start the publisher:
```
docker run --rm -it --network=host golang:1.13 sh -c "git clone https://github.com/aler9/goroslib && cd goroslib && go run examples/publisher.go"
```

Here I try to subscribe using rostopic
```
docker exec -it ros bash -c "source /opt/ros/melodic/setup.bash && rostopic echo /test_pub"
```

Here we can the type is being reported as `goroslib/Msg` insead of `sensor_msg/Imu`
```
docker exec -it ros bash -c "source /opt/ros/melodic/setup.bash && rostopic list -v"
```

I'm going to look into it myself (and possibly send you a pull request) but any insight you have would be appreciated. For an assignment I wrote an n-dimensional d* algorithm in Go that I planned to tie into ROS but I found a basic message didn't work. 

Thanks,

Jesse Stevenson

aler9, 2020-04-01:
Hi, when i was developing the library i discovered that the message name is totally useless, at least when using the C++/Python/Go/rosbridge+websocket client libraries: two clients are perfectly able to publish and subscribe to the same message, even if it has a different name on the two counterparts. Only the hash must match.

So i got rid of the message name (i always use `goroslib/msg`), in order to ease the deployment of this library.

Example: subscriber in C++
```
#include <ros/ros.h>

void on_message(const sensor_msgs::Imu::ConstPtr& in) {
}

int main(int argc, char **argv) {
    ros::init(argc, argv, "nodesub");
    ros::NodeHandle node("~");

    auto sub = node.subscribe("/test_pub", 100, on_message);

    ros::spin();
    return 0;
}
```

publisher in Go:
```
package main

import (
	"fmt"
	"time"

	"github.com/aler9/goroslib"
	"github.com/aler9/goroslib/msgs/geometry_msgs"
	"github.com/aler9/goroslib/msgs/sensor_msgs"
)

func main() {
	// create a node with given name and linked to given master.
	// master can be reached with an ip or hostname.
	n, err := goroslib.NewNode(goroslib.NodeConf{
		Name:       "/goroslib",
		MasterHost: "127.0.0.1",
	})
	if err != nil {
		panic(err)
	}
	defer n.Close()

	// create a publisher
	pub, err := goroslib.NewPublisher(goroslib.PublisherConf{
		Node:  n,
		Topic: "/test_pub",
		Msg:   &sensor_msgs.Imu{},
	})
	if err != nil {
		panic(err)
	}
	defer pub.Close()

	// publish a message every second
	ticker := time.NewTicker(1 * time.Second)
	for range ticker.C {
		msg := &sensor_msgs.Imu{
			AngularVelocity: geometry_msgs.Vector3{
				23.5,
				22.1,
				-7.5,
			},
		}
		fmt.Printf("Outgoing: %+v\n", msg)
		pub.Write(msg)
	}
}
```

But you're pointing out that in this way `rostopic echo` is unable to work, since it loads message definitions by name.
So i'll adjust the message name. Thanks.

aler9, 2020-04-04:
hello, this was fixed with commits from ee2ae23 to 90f7e66 .
now `rostopic echo` is fully supported, and these commands work:

```
docker run --rm -it --network=host --name=ros ros:melodic-ros-core-bionic roscore
docker run --rm -it --network=host golang:1.13 sh -c "git clone https://github.com/aler9/goroslib && cd goroslib && go run examples/publisher.go"
docker exec -it ros bash -c "source /opt/ros/melodic/setup.bash && rostopic echo /test_pub"
```

Let me know if there is any remaining problem.

closed 2020-04-04 by aler9