bluenviron/goroslib #4: Subscriber does not receive single message

romainreignier, 2020-04-30:
Hi!
I have discovered your work through the recent [Ubuntu blog post](https://ubuntu.com/blog/go-for-ros). Nice job!

You have done a great documentation job and I like how you test the issues in simple docker containers as in this comment https://github.com/aler9/goroslib/issues/2#issuecomment-601724672 :+1: 

My first try was to test a simple subscriber like the one in the examples and it seems to never receive a message from `rostopic pub`. But I have noticed that if I give a rate `-r 1`, the messages start to appear and from the `Seq: 1` so all messages seem to be received.

The issue is that sometimes, a node only waits for one message and not repeated messages so this behavior can be an issue.

Steps to reproduce:

1. roscore
    ```
    docker run --rm -it --network=host --name=roscore ros:melodic-ros-core-bionic roscore
    ```
2. Go subscriber
    ```
    docker run --rm -it --network=host golang:1.14 sh -c "git clone https://github.com/aler9/goroslib && cd goroslib && go run examples/subscriber.go"
    ```
3. `rostopic echo` to check that the message is published
    ```
    docker run --rm -it --network=host --name=rostopic_echo ros:melodic-ros-core-bionic rostopic echo /test_pub
    ```
4. `rostopic pub --once`
    ```
    docker run --rm -it --network=host --name=rostopic_pub ros:melodic-ros-core-bionic rostopic pub --once /test_pub sensor_msgs/Imu "header:
      seq: 0
      stamp: {secs: 0, nsecs: 0}
      frame_id: ''
    orientation: {x: 0.0, y: 0.0, z: 0.0, w: 0.0}
    orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
    angular_velocity: {x: 0.0, y: 0.0, z: 0.0}
    angular_velocity_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
    linear_acceleration: {x: 0.0, y: 0.0, z: 0.0}
    linear_acceleration_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]"
    ```

And here, I get no message from in the go subscriber console.

Now, if we publish with a rate, replace 4. with:

```
docker run --rm -it --network=host --name=rostopic_pub ros:melodic-ros-core-bionic rostopic pub -r 1 /test_pub sensor_msgs/Imu "header:
  seq: 0
  stamp: {secs: 0, nsecs: 0}
  frame_id: ''
orientation: {x: 0.0, y: 0.0, z: 0.0, w: 0.0}
orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: {x: 0.0, y: 0.0, z: 0.0}
angular_velocity_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
linear_acceleration: {x: 0.0, y: 0.0, z: 0.0}
linear_acceleration_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]"
```

It works!

aler9, 2020-05-02:
Hi, thanks for informing me about the blog post, i'm glad that many people found this library useful.
I decided to write it because i wanted to use ROS with JIT-compiled messages, since i worked with both ROS and grpc/protobuf, and they both required the compilation and sharing of message definitions between users. Of course i'd like to work on a ROS2 porting, and i'll start as soon as i'll find the time for it (in the meantime, any contribution is welcomed).

This issue is related to the latching feature of ROS, currently not implemented in the library, that consists in:
> When a connection is latched, the last message published is saved and automatically sent to any future subscribers that connect.

When a subscriber like `rostopic pub` wants to send a single message, this happens:
1. the subscriber informs the master that it is publishing on a given topic
2. the subscriber sends the message
3. the publisher gets informed by the master about the subscriber existence, and connects to it
4. since latching is not implemented, the publisher doesn't receive anything. Normally it would receive the last sent message from the subscriber.

So: i'll study how to implement latching.

romainreignier, 2020-05-02:
Ok, thanks for the explanation.
I think you have inverted *subscriber* and *publisher* terms in your scenario, isn't it?

aler9, 2020-05-02:
yes, thanks

closed 2020-05-02 by aler9

aler9, 2020-05-02:
Hi, this should be fixed,and i also added a couple of unit tests. Let me know if it works for you too. Bye

romainreignier, 2020-05-02:
Oh! That was quick!
Indeed, in my test case shown above, it is now working!
Thanks a lot!

commit b1eb9254 (2020-05-02): support latching; fix #4