bluenviron/goroslib #10: Support ROS Multimaster

aler9, 2020-05-17:


Filea, 2020-05-19:
When I try to subscribe something using goroslib while multimaster_fkie is running in background, I get the following error:

panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x10 pc=0x7fb1cf117133]goroutine 571 [running]:
github.com/aler9/goroslib.(*subscriberPublisher).run.func4(0xc000441440, 0xc000024190, 0xc000126a58, 0xc000d04190)
	github.com/aler9/goroslib@v0.0.0-20200511074020-4aad0d167d7c/subscriber-pub.go:105 +0x53
created by github.com/aler9/goroslib.(*subscriberPublisher).run
	github.com/aler9/goroslib@v0.0.0-20200511074020-4aad0d167d7c/subscriber-pub.go:103 +0x38b

aler9, 2020-05-24:
This has probably been fixed by c4b885bc: a couple of checks were being ignored, so there was a pointer with a null value.
Please let me know if there are more problems.

closed 2020-05-24 by aler9

Filea, 2020-05-25:
Good morning.

The commit that you refereed fixed the error that I was noticing. However, it seems that now, after a while, the subscriber stops receiving information. It happens after approximately 10 minutes. I tried with a simple string just to test and the problem persists. Interesting bug. 

Best, 
Filipe Pinto