bluenviron/goroslib #12: Desirability of basic type redefinitions

knuesel, 2020-05-23:
Hi, thanks for this package, it is extremely useful.

Is there a good reason to redefine the basic types in `msgs`? I have an application that translates ROS messages to similar protobuf messages. I have to cast every value (for example `msgs.Float64` to `float64`). And for slices I need to copy all values to new slices of the correct type. This wouldn't be necessary if goroslib was using the basic types (like `google.golang.org/protobuf`). Any chance you might agree to change it?

aler9, 2020-05-23:
Hi, the problem is that there's no one-to-one correspondence between all Go native types and ROS native types. As you can see in the code (https://github.com/aler9/goroslib/blob/master/msgs/base.go), a Go uint8 can correspond to a ROS byte, char or uint8, and this influences the message hash.

Since i can't pretend that every user knows this detail, i've chosen to use redefined types.

An alternative could be to use tags, like this:
```
type MessageName struct {
    msgs.Package  `ros:"my_package"`
    Field1 uint8       `ros:"byte"`
    Field2 int32       `ros:"int32"`
}```

But this would force users to write the field type twice.

What do you think?

knuesel, 2020-05-23:
I see... It's true that requiring tags on every field would be a step down in usability.

For `Package` vs `int`, there is a semantic difference, so I think it makes sense to have a `Package` type in Go.

This leaves `char`, `byte`, `uint8` and `int8`. According to http://wiki.ros.org/msg#Field_Types , the names `char` and `byte` are deprecated (by the way, it looks like there's a small bug in `msgs/base.go`: `type Byte uint8` should be `type Byte int8`). So what about using the basic Go types directly (meaning that ROS `uint8` and `int8` map to the Go types of the same name), and supporting the deprecated types `byte` and `char` through a special type (as currently done)?

closed 2020-05-26 by aler9

aler9, 2020-05-26:
native Go types are now used as basic types.
Char and byte have been intentionally left out, since they're deprecated. If someone will need them, i'll reimplement them with a tag-based system.

knuesel, 2020-05-30:
Perfect, thanks a lot!

commit 29fe8815 (2020-05-26): use native Go types for field types instead of custom types; fix #12