bluenviron/goroslib #13: Can't subscribe to CompressedImage topic

NikolaeVarius, 2020-05-24:
Using Melodic on Linux
go version go1.14.2 linux/arm64

```
        imgStream := make(chan *sensor_msgs.CompressedImage, 100)
	subTopic, err = goroslib.NewSubscriber(goroslib.SubscriberConf{
		Node:  n,
		Topic: "/output/image_raw/CompressedImage",
		Callback: func(msg *sensor_msgs.CompressedImage) {
			imgStream <- msg
		},
	})
	if err != nil {
		panic(err)
	}
	fmt.Println("Connected to Subscriber Topic")
	defer subTopic.Close()
```

```
Connected to Subscriber Topic
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x10 pc=0x6c52d4]

goroutine 30 [running]:
github.com/aler9/goroslib.(*subscriberPublisher).run.func4(0x400028e960, 0x400028a0d8, 0x4000210080, 0x400028c480)
	/home/jkim/go/src/github.com/aler9/goroslib/subscriber-pub.go:105 +0x4c
created by github.com/aler9/goroslib.(*subscriberPublisher).run
	/home/jkim/go/src/github.com/aler9/goroslib/subscriber-pub.go:103 +0x2a0
exit status 2

```

rqt_images shows the video frames rendering on that topic just fine.

Subscribing just to `/output/image_raw` does not panic, but nothing comes through on the topic

aler9, 2020-05-24:
Hi, thanks for the report, to fix this i need two things:
* can you try again with the latest version of the library ? the traceback reports an old version of `subscriber_pub.go`
* can you provide a way to reproduce the issue, for instance by writing the commands or the program you used to generate the stream?

thanks

NikolaeVarius, 2020-05-24:
Looks like the update fixed it, I'm receiving events now. 

Do you still want the original code that was used to publish? It was nothing special

closed 2020-05-25 by aler9