bluenviron/goroslib #14: Subscriber stops receiving messages when multimaster_fkie is running

Filea, 2020-05-25:
Good morning.

The commit that you refereed fixed the error that I was noticing when running multimaster_fkie and goroslib. However, it seems that now, after a while, the subscriber stops receiving information. It happens after approximately 10 minutes. I tried with a simple string just to test and the problem persists. Interesting bug.

Best,
Filipe Pinto

aler9, 2020-05-25:
Hi, thanks for the report, could you provide basic versions of the publisher and the subscriber?
Ideally i'd like to have a list of docker commands, like in this issue:
https://github.com/aler9/goroslib/issues/6

If docker commands are provided, i can then build an automated test to ensure this bug does not appear anymore.

Thanks

Filea, 2020-05-26:
Good morning. I'm not familiar with the Docker, but I can describe the steps I'm following. I hope it helps.
You just need to have two machines within the same network, multimaster_fkie installed on both machines and goroslib.

**Machine 1:**

- roslaunch master_discovery_fkie master_discovery.launch
- roslaunch master_sync_fkie master_sync.launch
- rostopic pub /test std_msgs/String "data: 'Hey'" -r 20

**Machine 2:**

- roslaunch master_discovery_fkie master_discovery.launch
- roslaunch master_sync_fkie master_sync.launch
- rostopic echo /test ( just to make sure that you can see what the other machine is sending)
- go build test.go
- ./test

The subscriber will stop receiving messages after approximately 10 minutes.

**GOROSLIB subscriber that I'm using:**

```
package main

import (
	"fmt"
	"time"

	"github.com/aler9/goroslib"
	"github.com/aler9/goroslib/msgs/std_msgs"
)

func onMessage(msg *std_msgs.String) {
	fmt.Printf("Incoming: %+v\n", msg)
}

func main() {
	time.Sleep(2 * time.Second)

	// create a node with given name and linked to given master.
	// master can be reached with an ip or hostname.
	n, err := goroslib.NewNode(goroslib.NodeConf{
		Name:       "/goroslib",
		MasterHost: "172.16.1.200",
	})
	if err != nil {
		panic(err)
	}
	defer n.Close()

	// create a subscriber
	sub, err := goroslib.NewSubscriber(goroslib.SubscriberConf{
		Node:     n,
		Topic:    "/test",
		Callback: onMessage,
	})
	if err != nil {
		panic(err)
	}
	defer sub.Close()

	// freeze main loop
	infty := make(chan int)
	<-infty
}
```

aler9, 2020-06-09:
Hi, i repeatedly tried to replicate this bug but without success. I'm able to establish a subscriber-publisher connection, with goroslib being the subscriber, between two separate machine with ros and ros multimaster running, for many hours.

I used:
* Ubuntu bionic (18.04) and ROS melodic
* Golang 1.14 and the last goroslib version

Both Ubuntu and Go are run in a Docker container, in order to use an environment that can be recreated with a couple of commands and can be replicated.

I'm posting the exact commands i've used below.

**First PC (publisher)**

```
docker run --rm -it --network=host \
ros:melodic-ros-core-bionic sh -c "\
apt update && apt install -y ros-melodic-multimaster-fkie \
&& (roscore &) \
&& sleep 3 \
&& (rosrun master_discovery_fkie master_discovery &) \
&& sleep 3 \
&& rosrun master_sync_fkie master_sync"
```
and
```
docker run --rm -it --network=host \
ros:melodic-ros-core-bionic sh -c "\
rostopic pub /test std_msgs/String \"data: 'Hey'\" -r 20"
```

**Second PC (subscriber)**

```
docker run --rm -it --network=host \
ros:melodic-ros-core-bionic sh -c "\
apt update && apt install -y ros-melodic-multimaster-fkie \
&& (roscore &) \
&& sleep 3 \
&& (rosrun master_discovery_fkie master_discovery &) \
&& sleep 3 \
&& rosrun master_sync_fkie master_sync"
```
and
```
docker build - -t temp << EOF
FROM golang:1.13
WORKDIR /s
RUN echo '\n\
module main\n\
require github.com/aler9/goroslib v0.0.0-20200608201439-746b429ea378\n\
' > go.mod
RUN echo '\n\
package main\n\
\n\
import (\n\
    "fmt"\n\
    "time"\n\
\n\
    "github.com/aler9/goroslib"\n\
    "github.com/aler9/goroslib/msgs/std_msgs"\n\
)\n\
\n\
func onMessage(msg *std_msgs.String) {\n\
    fmt.Printf("Incoming: %+v\\\n", msg)\n\
}\n\
\n\
func main() {\n\
    time.Sleep(2 * time.Second)\n\
\n\
    // create a node with given name and linked to given master.\n\
    // master can be reached with an ip or hostname.\n\
    n, err := goroslib.NewNode(goroslib.NodeConf{\n\
        Name:       "/goroslib",\n\
        MasterHost: "127.0.0.1",\n\
    })\n\
    if err != nil {\n\
        panic(err)\n\
    }\n\
    defer n.Close()\n\
\n\
    // create a subscriber\n\
    sub, err := goroslib.NewSubscriber(goroslib.SubscriberConf{\n\
        Node:     n,\n\
        Topic:    "/test",\n\
        Callback: onMessage,\n\
    })\n\
    if err != nil {\n\
        panic(err)\n\
    }\n\
    defer sub.Close()\n\
\n\
    // freeze main loop\n\
    infty := make(chan int)\n\
    <-infty\n\
}\n\
' > main.go
RUN go build -o /out .
ENTRYPOINT [ "/out" ]
EOF
docker run --rm -it --network=host temp
```

The test run smoothly, multiple times and for many hours.

Please check again with Go 1.14, Ubuntu 18.04, the last goroslib revision and let me know. Until i'm unable to replicate the bug, i cannot fix it.

Thanks for your time.

Filea, 2020-07-03:
Hello, I tested it again with the multimater_fkie default configuration and everything is running smoothly.
I think it was a conflict with the configuration that I was using. 

Thank you for your time.
Best, 
Filipe

closed 2020-07-03 by Filea