bluenviron/goroslib #16: Need help to use this lib

MatheusGodoi, 2020-07-02:
Followed the mini-tutorial in the README but I am sure I am missing something.
I have go 1.14 and ros kinetic installed, but I am unable to run any of the files for testing.

`go run node file.go` is not a valid command for me (go run node) and when I run just `go run file.go` I get a panic:
```
[signal SIGSEGV: segmentation violation code=0x1 addr=0x70 pc=0x689650]

goroutine 20 [running]:
github.com/aler9/goroslib.(*Publisher).Close(...)
	/home/claudio/go/src/github.com/aler9/goroslib/publisher.go:403
github.com/aler9/goroslib.(*Node).run(0xc0000da3c0)
	/home/claudio/go/src/github.com/aler9/goroslib/node.go:604 +0x1490
created by github.com/aler9/goroslib.NewNode
	/home/claudio/go/src/github.com/aler9/goroslib/node.go:347 +0x793
exit status 2
```

I feel like there should be something to run between `go mod init main` and `go run node file.go` or something to download before. Or maybe I messed up something without even knowing. Any help would be appreciated!

aler9, 2020-07-03:
Hi
* thanks for reporting that the `go run node file.go` is not a valid command, that was a typo this has been fixed, the right command is `go run file.go`
* thanks for reporting the segfault, this was indeed a bug that happens when the library cannot connect to the master. Now the library does not crash and an error message is returned.
* this problem happens when the library cannot connect to the master; please download the newer version of the library (it's enough to delete the go.mod file and recreate it with go mod init main), check the error and retry

MatheusGodoi, 2020-07-08:
Worked perfectly now, so I will close this.

Thanks for the quick help!

closed 2020-07-08 by MatheusGodoi