bluenviron/goroslib #22: md5sum check failed

JiHongzhou, 2020-10-16:
When i use this lib to pusblish costom msgs to another CPP node, The CPP node tell me the datatype/md5sum check failed, the log as bellow:

[ERROR] [1602764150.921571564]: Client [/avatar] wants topic /topic/response/signal/message to have datatype/md5sum [atris_msgs/SignalMessage/21bf35dedac23c2dd7049256f5f2f971], but our version has [atris_msgs/SignalMessage/47da1d8e018fea14775e1f7b4683149a]. Dropping connection.

JiHongzhou, 2020-10-16:
I find a mistake in msg/md5.go.

In the function md5Text, 

name := camelToSnake(ft.Name)

we convert name to Snake, but there are no rule to force all .msg to Snake.

aler9, 2020-10-16:
Hello, the situation is the following:
* all ROS standard messages have field names in snake case
* Go doesn't export fields with a name which begins with a letter in lower case; snake can't be used

Hence my idea was to convert field names from snake case into camel case, which is strongly recommended in Go and results in very clear structs; for example, this message
```
string name
geometry_msgs/Quaternion orientation
uint8 orientation_mode
uint8 interaction_mode
bool always_visible
visualization_msgs/Marker[] markers
bool independent_marker_orientation
string description
```

is converted into this
```go
type InteractiveMarkerControl struct {
	msg.Package                  `ros:"visualization_msgs"`
	Name                         string
	Orientation                  geometry_msgs.Quaternion
	OrientationMode              uint8
	InteractionMode              uint8
	AlwaysVisible                bool
	Markers                      []Marker
	IndependentMarkerOrientation bool
	Description                  string
}
```

The problem is that some custom messages are left out. Could you post a sample `.msg` file that doesn't work? in this way we can figure out together a solution. Thanks

JiHongzhou, 2020-10-19:
hmm, this message
```
string title
string ID
string msg
```
Maybe, this message is not fully compliant message，but ...

So I add a tag to fix this issue, 

```
type SignalMessage struct {
    msg.Package `ros:"my_package"`
    Title string `Title:"title"`
    MsgID string `MsgID:"ID"`
    Msg string `Msg:"msg"`
}
```

JiHongzhou, 2020-10-19:
Read the tag first, if no tag, use default.
```
--- a/msg/md5.go
+++ b/msg/md5.go
@@ -140,7 +140,10 @@ func md5Text(rt reflect.Type, rosTag string) (string, bool, error) {
                        }
 
                        //name := camelToSnake(ft.Name)
-                       name := ft.Tag.Get("typename")
+                       name := ft.Tag.Get(ft.Name)
+                       if name == "" {
+                               name = camelToSnake(ft.Name)
+                       }
```

aler9, 2020-11-03:
Field names not in snake case are now fully supported; the name can be set with the `rosname` tag:
```go
type MessageName struct {
    msg.Package `ros:"my_package"`
    Field bool  `rosname:"FIELD"`
}
```

closed 2020-11-03 by aler9