bluenviron/goroslib #25: Namespacing issues

timanema, 2020-11-02:
Let me begin by saying I'm a beginner with ROS, so if this issue is just because of something I'm doing wrong, let me know. If I missed something in the docs, or this is just not supported, also let me know :)

### The problem
I was unable to find something about namespaces, so I just assumed we are supposed to do this by having an appropriate node name. For example: `/namespace/node`. It looks like this is working, as the ros master will successfully remap new subscribers/publishers to the namespace. So creating a subscriber which subscribes to `/topic` on a node called `/namespace/node`, will successfully subscribe to `/namespace/topic`, and the same goes for a publisher.

Unfortunately, goroslib doesn't seem to handle namespaces that well. When using namespaces, a `goroslib.Subscriber` subscribed to `/topic` (which will be remapped to `/namespace/topic`), will successfully receive messages from already existing publishers. However, this is not the case for publishers created after said subscriber.

To demonstrate this I have written a small test:
```golang
package main

import (
	"fmt"
	"time"
)

func main() {
	// No namespace
	test("")

	// Namespace
	test("namespace/")
}

func test(namespace string) {
	fmt.Printf("Using namespace '%s'\n", namespace)

	// Create publisher before the subscriber, this should work
	p1 := NewPub(fmt.Sprintf("/%spub1", namespace))

	// Create the actual subscriber
	s := NewSub(fmt.Sprintf("/%ssub", namespace))

	// Create a publisher after the subscriber is running, the subscriber will not receive updates from this one
	p2 := NewPub(fmt.Sprintf("/%spub2", namespace))

	time.Sleep(time.Second * 2)

	p1.Publisher.Write(&Message{
		Msg:     "Msg before sub",
	})
	p2.Publisher.Write(&Message{
		Msg:     "Msg after sub",
	})

	time.Sleep(time.Second)
	s.Close()
	p1.Close()
	p2.Close()
	fmt.Printf("---\n")
}
```
<details>
  <summary>Other boilerplate code</summary>

```golang
package main

import (
	"github.com/aler9/goroslib"
)

// Creates a publisher on the '/topic' topic.
type Pub struct {
	Node      *goroslib.Node
	Publisher *goroslib.Publisher
}

func NewPub(name string) *Pub {
	p := &Pub{}
	var err error

	p.Node, err = goroslib.NewNode(goroslib.NodeConf{
		MasterHost: "localhost",
		MasterPort: 11311,
		Name:       name,
	})
	if err != nil {
		panic(err)
	}

	p.Publisher, err = goroslib.NewPublisher(goroslib.PublisherConf{
		Node:  p.Node,
		Topic: "/topic",
		Msg:   &Message{},
	})

	return p
}

func (p *Pub) Close() {
	if err := p.Publisher.Close(); err != nil {
		panic(err)
	}
	if err := p.Node.Close(); err != nil {
		panic(err)
	}
}
```
```golang
/// main.go
package main

import (
	"fmt"
	"github.com/aler9/goroslib"
)

// Creates a subscriber for the '/topic' topic, which just prints received messages.
type Sub struct {
	Node       *goroslib.Node
	Subscriber *goroslib.Subscriber
}

func NewSub(name string) *Sub {
	s := &Sub{}
	var err error

	s.Node, err = goroslib.NewNode(goroslib.NodeConf{
		MasterHost: "localhost",
		MasterPort: 11311,
		Name:       name,
	})
	if err != nil {
		panic(err)
	}

	s.Subscriber, err = goroslib.NewSubscriber(goroslib.SubscriberConf{
		Node:     s.Node,
		Topic:    "/topic",
		Callback: s.callback,
	})

	return s
}

func (s *Sub) callback(msg *Message) {
	fmt.Printf("sub msg: %v\n", msg)
}

func (s *Sub) Close() {
	if err := s.Subscriber.Close(); err != nil {
		panic(err)
	}
	if err := s.Node.Close(); err != nil {
		panic(err)
	}
}
```
```golang
package main

import (
	"github.com/aler9/goroslib/msg"
)

type Message struct {
	msg.Package `ros:"ras"`
	Msg         string
}

```

</details>

The output is:
```
Using namespace ''
sub msg: &{0 Msg before sub}
sub msg: &{0 Msg after sub}
---
Using namespace 'namespace/'
sub msg: &{0 Msg before sub}
---
```
As you can see, the subscriber does indeed not receive updates from the publisher which was created after the subscriber.

#### Reproducing
The code can be found in the previous section.
I used ros noetic (1.15.8) in docker for testing: 
`docker run -it --network=host ros:noetic-ros-core roscore`

### Possible cause
Before making this issue, I did some digging, and I think I may have found the cause. When a new subscriber is created in node.go, it registers it with the master, and receives the existing publishers. This is why existing publishers do work:
```golang
// node.go:440
res, err := n.apiMasterClient.RegisterSubscriber(apimaster.RequestRegister{
				Topic:     req.sub.conf.Topic[1:],
				TopicType: req.sub.msgType,
				CallerUrl: n.apiSlaveServerUrl,
			})
			if err != nil {
				req.err <- err
				continue
			}

			n.subscribers[req.sub.conf.Topic] = req.sub
			req.err <- nil
```
In the test the relevant values will be:
* Node name (n.conf.Name): `/namespace/sub`
* Topic (req.sub.conf.Topic): `/topic`
* Status response (res.StatusMessage): `Subscribed to [/namespace/topic]`

This is another confirmation that the master does the remapping. 
  
Whenever a new publisher is created, node.go will handle the updating of urls. However, the topic check is done with the original topic, not with the remapped one. Since the ros master will send the remapped version, this check will fail, even though the publisher is in the same namespace as the subscriber:
```golang
//node.go:404
case req := <-n.publisherUpdate:
			sub, ok := n.subscribers[req.topic]
			if !ok {
				continue
			}

			sub.publisherUpdate <- req.urls
```
In the test the relevant values will be:
* Received topic (req.topic): `/namespace/topic`
* Subscribers (n.subscribers): `{"/topic": [...]}`

This shows the mismatch between the original and remapped topics.

### Possible solutions
A quick workaround for now is to put everything in the global namespace, although this is not a solution for bigger projects or projects with conflicting node names. I think a complete solution will have to be compliant with the [ROS Names](http://wiki.ros.org/Names), so it might take some time to fully implement. 
Something quick and dirty that could work for most cases would be something like [this](https://play.golang.org/p/0fG2auJwC2x). In this solution you would just create a list of allowed namespaces, append the original topic name, and check for any match.

timanema, 2020-11-02:
I haven't tested this with services, but something similar might be happening there. So in case this really is an issue, services also might need a second look.

aler9, 2020-11-22:
Hello, thank you very much for reporting this bug. I finally got time to check and is true that the master, when it receives a request through the Master API, prepends the node namespace to all topics that doesn't begin with a slash, so a call like this, performed by a node `/namespace/node`

```go
topic := "/mytopic"
n.apiMasterClient.RegisterSubscriber(apimaster.RequestRegister{
  Topic:     topic[1:],
  TopicType: "mytype",
  CallerUrl: "http://url",
})
```

will result in subscribing the topic `/namespace/mytopic` instead of `/mytopic`.

You may ask: why is the slash dropped before performing the request (ie `topic[1:]`)? because the reference c++ implementation does it, and since tests were successful, it was left in this way, until you discovered the bug.

In my opinion there are two problems that must be solved separately:
* allow nodes not belonging to a namespace to correctly subscribe or publish to topics with multiple slashes
* support for namespaces, that leads to supporting relative and private topic names

i'd start with the first one (which is a bug), and in my opinion the best solution consists in passing full topic names (with the initial slash) when communicating with the master, even if this is not the solution enrolled by the official reference.

i've already done some tests and it seems to work well.

we can then proceed to implement namespaces and support relative and private topic names.

aler9, 2020-11-23:
namespaces are not fully supported; a node namespace can be set with the Namespace member:

```go
NewNode(NodeConf{
    Namespace:  "/myns",
    Name:       "goroslib",
    MasterHost: masterHost,
})
```

closed 2020-11-23 by aler9