bluenviron/goroslib #33: Unable to receive any new messages

derektan95, 2020-12-07:
Hi, 
Firstly, thank you for creating such a useful library. I would like to express some of the issues I am currently facing when using this library. Currently, I have a ROS environment (ROS Kinetic on Ubuntu 16.04) on my robot subscribing to a topic called '/cmd_vel'. On the server end, I have a Windows environment with Golang (Windows 10 Pro, Go 1.15.3), responsible for publishing the topic '/cmd_vel'. 

I firstly did a roscore on the client side (robot), with the IP of 192.168.123.161 and port of 11311. I edited the following script such that this Golang script publishes 'geometry_msgs.Twist' message on the server side to the robot. Content published is displayed on the terminal. However on the client side, I can see the topic being published when i did `rostopic list -v`, but when I performed a `rostopic echo /cmd_vel`, it is an empty message. `rostopic hz /cmd_vel` indicates that no new mesage is coming in, which implies that the initial 'handshake' is established but somehow messages are being dropped. Additionally, the published topic does not go away in `rostopic echo /cmd_vel` even when I had stopped publishing from server side. 

When I attempted to subscribe to another topic from server (Golang), there is no new message received as well. Here is the code I executed on Windows. Let me know if you have any insights to this issue, thanks! 

`
func RosSending() {
	// create a node with given name and linked to given master.
	n, err := goroslib.NewNode(goroslib.NodeConf{
		Name:          "goroslib_pub",
		MasterAddress: "192.168.123.161:11311",
	})
	if err != nil {
		panic(err)
	}
	defer n.Close()

	// create a publisher
	pub, err := goroslib.NewPublisher(goroslib.PublisherConf{
		Node:  n,
		Topic: "/cmd_vel",
		Msg:   &geometry_msgs.Twist{},
	})
	if err != nil {
		panic(err)
	}
	defer pub.Close()

	// publish a message every second
	ticker := time.NewTicker(2 * time.Millisecond)
	defer ticker.Stop()

	for range ticker.C {
		msg := &geometry_msgs.Twist{
			Linear: geometry_msgs.Vector3{
				X: x,
				Y: y,
				Z: z,
			},
			Angular: geometry_msgs.Vector3{
				X: ax,
				Y: ay,
				Z: az,
			},
		}
		fmt.Printf("Outgoing: %+v\n", msg)
		pub.Write(msg)
	}
}
`

aler9, 2020-12-07:
Hello, i tried to replicate the issue, but in my case everything worked flawlessly. I did the following:

1. on the first pc, which is running Ubuntu 20.04 and has ROS Noetic installed, i started the master
    ```
    roscore
    ```

2. on the 2nd pc, which is running Windows 10 20H2, and has Go 1.15.3 installed, i downloaded your script and started it
    ```
    go run main.go
    ```

3. on the first pc, i started rostopic echo
    ```
    rostopic echo /cmd_vel
    ```

and i was able to correctly see the periodic messages
```
---
linear: 
  x: 1.0
  y: 2.0
  z: 3.0
angular: 
  x: 4.0
  y: 5.0
  z: 6.0
---
```

In my opinion, it could be a firewall problem, since
* rostopic list requires a connection with the master only, not with the publisher
* rostopic echo requires a connection with the publisher

so try disabling firewall on both sides.

derektan95, 2020-12-16:
Hi @aler9 ,

Thank you so much for the detailed explanation. I should have clarified this earlier; we actually used an Ubuntu virtual machine (Ubuntu 16.04) hosted on Windows via Virtual Box. This is used to communicate with the windows computer with Golang installed. 

Unfortunately, we attempted to disable firewalls on both the server and client sides (on both windows platform and the Ubuntu VM), but the issue persisted. Would you think this issue is related to the VM? We are sure we the Windows environment with Golang could ping the ubuntu VM.

derektan95, 2020-12-23:
Hi @aler9,

I retested this setup on Ubuntu 18.04 on ROS Melodic and both publishing and subscribing with GoRosLib works. Thanks for your help!

closed 2020-12-23 by derektan95