bluenviron/goroslib #36: Support of ROS2

xo01, 2020-12-30:
Does this library support ROS2? if not then any plans to support ROS2.

Thanks in advance

aler9, 2020-12-30:
Absolutely YES but first i need to implement the [missing features](https://github.com/aler9/goroslib/blob/master/FEATURES.md).

bartekpacia, 2021-02-14:
I'd love to see this awesome library getting support for ROS2 as well!

aler9, 2021-02-14:
The main problem of ROS2 is that it's based on another, more generic, protocol, that is DDS, that is relatively newer. There isn't a Go implementation, and it must be build from scratch. This is long and expensive, but i hope it will be done at some point.
After it, a pure-Go ROS2 implementation can be set up.

https://en.wikipedia.org/wiki/Data_Distribution_Service
https://www.omg.org/spec/DDS/About-DDS/

pvandervelde, 2021-09-04:
Any progress on getting ROS2 support? I'd love to use Golang instead of Python

pvandervelde, 2022-01-05:
> The main problem of ROS2 is that it's based on another, more generic, protocol, that is DDS, that is relatively newer. There isn't a Go implementation, and it must be build from scratch. This is long and expensive, but i hope it will be done at some point. After this step, a pure-Go ROS2 implementation can be developed.
> 
> https://en.wikipedia.org/wiki/Data_Distribution_Service https://www.omg.org/spec/DDS/About-DDS/ https://design.ros2.org/articles/ros_on_dds.html

Is this library useful for handling the DDS  part: https://github.com/rticommunity/rticonnextdds-connector-go?

l1va, 2022-01-12:
Still too young, but can be better than nothing.

erichuiyliu, 2022-12-09:
Hey, any updates to support ROS2? Looking forward to it~~

LarryDCJ, 2023-01-15:
Is anyone working on this feature? I'd be happy to help out where I can.

LarryDCJ, 2023-02-08:
> Is anyone working on this feature? I'd be happy to help out where I can.

I’m going to start working on the DDS implementation. @aler9 I’d love to confer with you during the course of development.

aler9, 2023-02-08:
@LarryDCJ hello,

First of all, this library is practically in maintenance mode, since all ROS1 features have been implemented a while ago. Recent commits have been about updating dependencies or bug fixing.

The reason why a Golang ROS2 implementation was not developed yet is because there are not enough reasons to do it. Personally i can't use ROS in my job since its role as middleware is already taken by Kafka, AMQP or SQS, all of which can be horizontally scaled and have a better handling of undelivered messages, while its role as message encoder is already taken by gRPC. The time i can spend on OSS is currently focused on [some other](https://github.com/aler9/rtsp-simple-server) protocols which are way more popular.

Furthermore, this porting was never recognized by the official organization (although it's one of the most popular ones), so until now no one showed up offering concrete help.

If someone is willing to start a DDS implementation, i'm willing to review the work, but i can't do more than that at the moment.

LarryDCJ, 2023-02-08:
@aler9 Thanks for the insight!

I'm currently writing a driver for the [YDLidar G2](https://www.ydlidar.com/Public/upload/files/2022-07-25/YDLIDAR%20G2%20Development%20Manual%20V1.1(211230).pdf) and wanted to implement Ros2 written in Go when it's all done. 

This is all super new to me as I'm just learning to work with serial data.

ppbrown, 2023-08-12:
Did you ever look at
https://github.com/yangchenxing/go-dds
?
It's quite old, so I would expect you found it before, but... figured I'd ask.

jbcpollak, 2023-09-12:
For anyone who's wants to give a Golang ROS2 interface a shot, rather than native golang all the way to the DDS layerz interfacing Go to ROS2's rcl and rmw libraries via C might be a better path. This is how the Python and JavaScript implementions work, and allows the language interface to be independent of the actual middleware that is used, which is good because while DDS is the default, there are other middleware implementations.

The nodejs implementation may be useful as a starting point:

https://github.com/RobotWebTools/rclnodejs

bchoineubility, 2023-12-19:
Hi,

Does this library support ros2-galactic ? do you have plan for ros2-humble as well?

BR,
Mark

mari-muthu-k, 2024-04-26:
Hi, Does this library support ros2-humble ?

mari-muthu-k, 2025-04-19:
Hi @leoxiang66 , I guess No updates from them yet.