bluenviron/goroslib #37: Error with custom srv

gwaxG, 2021-02-01:
Hello,
I wanted to use this package in my work, but I met a problem with custom services.
When I launch the example "serviceprovider-custom.go" and try to call this service through the CLI, I get this:
"ERROR: Unable to load type [my_package/TestService].
Have you typed 'make' in [my_package]?"
The error is the same for custom messages.
How can it be solved?
I would appreciate any help!

aler9, 2021-02-02:
Hello, if you want to use cli (i.e. `rosservice call`) with custom messages, written with Go, Python or C, you have to pass the message definition to the cli.

You have to write the `.srv` equivalent of the Go structure, that in case of `serviceprovider-custom.go` is

```
float64 a
string b
---
float64 c
```

Then you have to insert it into a standard ROS project and compile it with cmake.

Finally, you have to load the project with `source workspace/install/setup.bash`.

Then, you can finally call `rosservice`.

gwaxG, 2021-02-03:
Thank you for your answer!
It was not clear how it works if we want to use Go nodes which do not need message and service definitions. 
I guess it is  also necessary to point `msg.Package ros:"my_pacakge"` to the correct package.

closed 2021-02-03 by gwaxG