bluenviron/goroslib #40: Port assignement with roslaunch

gwaxG, 2021-03-26:
Hello,
I have an issue when I try to launch Go nodes with roslaunch.
Basically, when I use a construction like `roslaunch test test.launch`, everything works fine, but
when I define a port like `roslaunch -p 11315 test test.launch`, the Go node crushes with the error:
`unable to solve master address: lookup tcp/11315/: Servname not supported for ai_socktype
[monitor_app-2] process has died [pid 54703, exit code 1, cmd .....`, meanwhile python nodes work well.
I have figured out a particular solution. If I add a line inside a Go node `os.Setenv("ROS_MASTER_URI", "http://localhost:11315")` before node creation, the Go node starts working without errors.
Is it possible to make Go nodes behaving like python ones?

aler9, 2021-03-27:
Hello, sorry but i don't understand - goroslib is configured with structs, not with environment variables, and is not launched with roslaunch (unless you've found a way to do it) - can you post some sample code to replicate the issue? thanks

gwaxG, 2021-05-26:
Hello, I launch Go nodes through roslaunch. It works in the same way as for C++, binary files could be launched after compilation. 
I have found my point of misunderstanding considering environment variables, sorry for bothering you.

closed 2021-05-26 by gwaxG