bluenviron/goroslib #45: Improper String Constant Handling when using msg-import Command

dxmaxwell, 2021-04-14:
## Which operating system are you using?

OS

- [X] Linux
- [ ] Windows
- [ ] macOS

Architecture

- [X] amd64
- [ ] arm64
- [ ] arm7
- [ ] arm6

## Description

When using the `msg-import` command constants of type **string** are not handled property.

For example ROS definition:
```
# A simple ROS message definition
string CONSTANT1=CONSTANT_VALUE_1

string source
```
The following **invalid** Go code is generated:
```
package main //nolint:golint

import (
    "github.com/aler9/goroslib/pkg/msg"
)

const (
    Example1_CONSTANT1 string = CONSTANT_VALUE_1 //nolint:golint
)

type Example1 struct { //nolint:golint
    msg.Package `ros:"ocu_msgs"`
    msg.Definitions `ros:"string CONSTANT1=CONSTANT_VALUE_1"`
    Source string//nolint:golint
}
```
The initialization value for `Example1_CONSTANT1` is missing quotations, it should be `"CONSTANT_VALUE_1"`

A separate but perhaps related issue, for this example ROS definition file:
```
# A simple ROS message definition
string CONSTANT1="CONSTANT_VALUE_1"

string source
```
The following Go code is generated:
```
package main //nolint:golint

import (
    "github.com/aler9/goroslib/pkg/msg"
)

const (
    Example2_CONSTANT1 string = "CONSTANT_VALUE_1" //nolint:golint
)

type Example2 struct { //nolint:golint
    msg.Package `ros:"ocu_msgs"`
    msg.Definitions `ros:"string CONSTANT1="CONSTANT_VALUE_1""`
    Source string//nolint:golint
}
```
The problem here is more subtle, the struct tag for `msg.Definitions` is invalid because there are unescaped quotation marks. Instead is should be:
```
msg.Definitions `ros:"string CONSTANT1=\"CONSTANT_VALUE_1\""`
```

aler9, 2021-04-14:
Hello, thanks for reporting the issue, the problem has been fixed and unit tests have been added.

closed 2021-04-14 by aler9

aler9, 2021-04-15:
I did some additional research, it turned out that this definition

```
# A simple ROS message definition
string CONSTANT1="CONSTANT_VALUE_1"

string source
```

means that CONSTANT1 starts with a quote character and ends with a quote character. Therefore, the proper Go code equivalent to definition is

```go
package main 

import (
    "github.com/aler9/goroslib/pkg/msg"
)

const (
    Example2_CONSTANT1 string = "\"CONSTANT_VALUE_1\""
)

type Example2 struct { 
    msg.Package `ros:"ocu_msgs"`
    msg.Definitions `ros:"string CONSTANT1=\"CONSTANT_VALUE_1\""`
    Source string
}
```

This scenario is now also supported.

dxmaxwell, 2021-04-15:
Thank you for addressing this issue so quickly.  ROS string handling is bit strange, but it looks correct to me now!

However, I'm having another error with the latest version:
```
Example1.go:5:5: imported and not used: "github.com/aler9/goroslib/pkg/msgproc"
Example1.go:14:5: undefined: msg
Example1.go:15:5: undefined: msg
```

aler9, 2021-04-15:
Replace "msgproc" with "msg" in your example file

dxmaxwell, 2021-04-15:
Not sure I understand, from this msg file:
```
# A simple ROS message definition
string CONSTANT1=CONSTANT_VALUE_1

string source
```

the following Go is generated:
```
//nolint:golint
package main

import (
    "github.com/aler9/goroslib/pkg/msgproc"
)

const (
    Example1_CONSTANT1 string = "CONSTANT_VALUE_1"
)

type Example1 struct {
    msg.Package `ros:"ocu_msgs"`
    msg.Definitions `ros:"string CONSTANT1=CONSTANT_VALUE_1"`
    Source string
}
```
Which seems to have the wrong import statement.

Do I need to manually change the generated file now?

aler9, 2021-04-15:
Sorry, i didn't understood that the code was generated by msg-import, and unit tests were not covering imports - this is fixed too.