bluenviron/goroslib #52: Simple action client stuck waiting for server

lazaroofarrill, 2021-06-03:
## Which operating system are you using?

OS

- [x] Linux
- [ ] Windows
- [ ] macOS

Architecture

- [x] amd64
- [ ] arm64
- [ ] arm7
- [ ] arm6

## Describe the issue

Description
I just updated, now the action clients are stuck in WaitForServer()

aler9, 2021-06-03:
Hello, the `WaitForServer` algorithm was not changed at all. Please try again, or alternatively provide a way to replicate the issue, for instance:

1) start the master node
2) run action server... (provide code or command line)
3) run action client...

lazaroofarrill, 2021-06-04:
Hello, I rolled back my code and it's working so I guess I messed up somewhere, sorry.

closed 2021-06-04 by lazaroofarrill

sniperZj, 2021-08-24:
It happened to me, too
code:
  node, err := goroslib.NewNode(goroslib.NodeConf{
		  Name:          "xx_node",
		  Host:          "127.0.0.0.1",
		  MasterAddress: "127.0.0.0.1:11311",
	  })
	  if err != nil {
		  panic(err)
	  }
	  defer node.Close()
	  sac, err := goroslib.NewSimpleActionClient(goroslib.SimpleActionClientConf{
		  Node:   node,
		  Name:   "robot_pose_editor",
		  Action: &action.PoseEditAction{},
	  })
  
	  if err != nil {
		  c.JSON(http.StatusOK, gin.H{
			  "code":    200,
			  "message": err.Error(),
			  "pose_id": 0,
		  })
		  return
	  }
  
	  defer sac.Close()
	  // wait for the server
	  fmt.Println("wait for the server")
	  sac.WaitForServer()
	  fmt.Println("create a simple action client end")
           .........
  -------------------------------------------------------
describe ：
            failed to print properly "create a simple action client end"
            rostopic echo robot_pose_editor/status
            header: 
              seq: 4332
              stamp: 
                secs: 1629807103
                nsecs: 470941966
              frame_id: ''
            status_list: []
            feedback、result、goal、cancel topic not data

sniperZj, 2021-08-24:
@aler9 Looking forward to your reply. Thank you @