bluenviron/goroslib #55: TCP Protocol Error with Publisher Topic Check ("wrong topic")

dxmaxwell, 2021-07-08:
## Which operating system are you using?

OS

- [X] Linux
- [ ] Windows
- [ ] macOS

Architecture

- [X] amd64
- [ ] arm64
- [ ] arm7
- [ ] arm6

## Describe the issue

When subscribed to a topic no messages are received.
Note: the topic has been verified to be actively publishing using the `rostopic echo /topic` command.

Issue 1: The subscriber silently ignores errors that are returned during execution of the TCP protocol:
See: https://github.com/aler9/goroslib/blob/499e979f2f583a2fbffad8ee6773b4524a41c9bf/subscriberpub.go#L60-L63

This made if much more difficult to identify the actual problem. I've had a brief look at the code and it looks like there is no clear way to report these errors. (Hoping you may have some ideas?) However, it would be good if these could at least be logged.

Issue 2: The TCP protocol `runInnerTCP` returns a "wrong topic" error.  In our case the publisher of the topic does not does not include the topic in their reply header. (As it is not strictly required by the spec: http://wiki.ros.org/ROS/TCPROS) 

See: https://github.com/aler9/goroslib/blob/499e979f2f583a2fbffad8ee6773b4524a41c9bf/subscriberpub.go#L242-L244

If the response from the publisher does not include the topic, then this check should not be done, potentially fixed by checking if the topic is empty, for example:
```
if outHeader.Topic != "" && outHeader.Topic != sp.sub.conf.Node.absoluteTopicName(sp.sub.conf.Topic) {
   return fmt.Errorf("wrong topic")
}
```

In addition it would be good to add a check for message type (along with the MD5 sum which you already have), which are both required by the spec.

This publisher causing this error uses is a custom ROS implementation that follows the spec more closely then the standard/official ROS tools and libraries.

aler9, 2021-07-25:
Hello,

> However, it would be good if these could at least be logged.

This library was designed with the aim of being as similar as possible to the official C++ implementation: entities (nodes, subscribers, publishers) and callbacks are pretty much the same. In the standard library, errors and warnings are reported through logging, that here is currently missing.

Therefore, the issue can be solved in two ways: by implementing logging and printing log lines to the log topic and to stdout, or by adding a custom callback that allows to parse errors. Probably the log way is the best option, but i have to dig further into the behavior of the standard library.

> If the response from the publisher does not include the topic, then this check should not be done, potentially fixed by checking if the topic is empty

This can be implemented.

> it would be good to add a check for message type

The problem of this change is that messages sometimes belongs to different namespaces from library to library, and this influences the name of the message type. At the same time, the name of the message type doesn't influence anything other than a field in the handshake. I'd rather not implement it.

dxmaxwell, 2021-07-27:
Thank you for looking into this issue.  In terms of logging I think it is really nice when a library provides an interface that allows log messages from the library to be directed into the logging framework of the application (ie Echo Logger, Logrus, etc.).  Since the standard library does not provide a standard logging interface, I have used this simple pattern in a few small libraries with success:
(For example add logging.go in the goroslib root)
```
package goroslib

import "log"

// Logger is a general interface that is compatible with Echo Logger and Logrus
type Logger interface {
	Debug(args ...interface{})
	Debugf(format string, args ...interface{})

	Info(args ...interface{})
	Infof(format string, args ...interface{})

	Warn(args ...interface{})
	Warnf(format string, args ...interface{})

	Error(args ...interface{})
	Errorf(format string, args ...interface{})
}

// StdLogger is an implementation of the Logger interface using the default standard log
type StdLogger struct{}

// Debug logs a message at the DEBUG level
func (l *StdLogger) Debug(args ...interface{}) {
	debugArgs := make([]interface{}, 0, len(args)+1)
	debugArgs = append(debugArgs, "[DEBUG]")
	debugArgs = append(debugArgs, args...)
	log.Print(debugArgs...)
}

// Debugf logs a formatted message at the DEBUG level
func (l *StdLogger) Debugf(format string, args ...interface{}) {
	log.Printf("[DEBUG] "+format, args...)
}

// Info logs a message at the INFO level
func (l *StdLogger) Info(args ...interface{}) {
	infoArgs := make([]interface{}, 0, len(args)+1)
	infoArgs = append(infoArgs, "[INFO]")
	infoArgs = append(infoArgs, args...)
	log.Print(infoArgs...)
}

// Infof logs a formatted message at the INFO level
func (l *StdLogger) Infof(format string, args ...interface{}) {
	log.Printf("[INFO] "+format, args...)
}

// Warn logs a message at the WARN level
func (l *StdLogger) Warn(args ...interface{}) {
	warnArgs := make([]interface{}, 0, len(args)+1)
	warnArgs = append(warnArgs, "[WARN]")
	warnArgs = append(warnArgs, args...)
	log.Print(warnArgs...)
}

// Warnf logs a formatted message at the WARN level
func (l *StdLogger) Warnf(format string, args ...interface{}) {
	log.Printf("[WARN] "+format, args...)
}

// Error logs a message at the ERROR level
func (l *StdLogger) Error(args ...interface{}) {
	errorArgs := make([]interface{}, 0, len(args)+1)
	errorArgs = append(errorArgs, "[ERROR]")
	errorArgs = append(errorArgs, args...)
	log.Print(errorArgs...)
}

// Errorf logs a formatted message at the ERROR level
func (l *StdLogger) Errorf(format string, args ...interface{}) {
	log.Printf("[ERROR] "+format, args...)
}

// Log is the Logger instance used by this library
var Log Logger = &StdLogger{}
```

You may have a better way to achieve this result, but I recommend it over directly printing to StdErr.

aler9, 2021-09-08:
A log system has been implemented; the behavior is the same of the reference C++ implementation, log messages are printed to stdout/stderr depending on their severity, and also sent to `/rosout`, that can be inspected with `rqt_console`.

closed 2021-09-08 by aler9