bluenviron/goroslib #60: Topics aren't removed from the ROS master topic list on normal node Shutdown

Mohamedemad4, 2021-07-25:
## Which operating system are you using?

OS

- [x] Linux
- [ ] Windows
- [ ] macOS

Architecture

- [x] amd64
- [ ] arm64
- [ ] arm7
- [ ] arm6

## Describe the issue
Topics aren't removed from the ROS master topic list on normal node Shutdown (SIGTERM)
this also applies to things like node name.

aler9, 2021-09-08:
Hello, i checked and it's true that the reference C++ library does listen on SIGTERM and removes topics from the master. Probably it's better to implement this feature in Go too instead of relying on defer().

aler9, 2021-10-01:
All the examples have been upgraded to correctly handle CTRL-C / SIGINT / SIGTERM, and shutdown the node thereafter.
In this way, users can choose to listen on OS signals like the standard library, or not to, since it causes quite a mess in case of complex programs with multiple objects.

closed 2021-10-01 by aler9