bluenviron/goroslib #80: Simple action client WaitForServer Has been waiting for

aler9, 2021-09-11:
### Discussed in https://github.com/aler9/goroslib/discussions/69

<div type='discussions-op-text'>

<sup>Originally posted by **sniperZj** August 24, 2021</sup>
code:
  node, err := goroslib.NewNode(goroslib.NodeConf{
		  Name:          "xx_node",
		  Host:          "127.0.0.0.1",
		  MasterAddress: "127.0.0.0.1:11311",
	  })
	  if err != nil {
		  panic(err)
	  }
	  defer node.Close()
	  sac, err := goroslib.NewSimpleActionClient(goroslib.SimpleActionClientConf{
		  Node:   node,
		  Name:   "robot_pose_editor",
		  Action: &action.PoseEditAction{},
	  })
  
	  if err != nil {
		  c.JSON(http.StatusOK, gin.H{
			  "code":    200,
			  "message": err.Error(),
			  "pose_id": 0,
		  })
		  return
	  }
  
	  defer sac.Close()
	  // wait for the server
	  fmt.Println("wait for the server")
	  sac.WaitForServer()
	  fmt.Println("create a simple action client end")
           .........
  -------------------------------------------------------
failed to print properly "create a simple action client end"
rostopic echo robot_pose_editor/status
header: 
  seq: 4332
  stamp: 
    secs: 1629807103
    nsecs: 470941966
  frame_id: ''
status_list: []
feedback、result、goal、cancel topic not data</div>

aler9, 2021-09-11:
@sniperZj please set `LogVerbosity` of `NodeConf` to `LogLevelDebug` and post any log line that appears on the console.

sniperZj, 2021-09-11:
@aler9 Sorry, it's my problem, it can be closed

closed 2021-09-11 by aler9