bluenviron/goroslib #94: Missing `message_definition` in publisher header

Edholm, 2021-11-10:
## Which operating system are you using?

OS

- [x] Linux
- [ ] Windows
- [ ] macOS

Architecture

- [x] amd64
- [ ] arm64
- [ ] arm7
- [ ] arm6

## Describe the issue

Description

It seems that the `message_definition` header is "missing" from the `HeaderPublisher`. From what I can gather from [here](http://wiki.ros.org/ROS/TCPROS) it shouldn't be needed (which seems weird to me...) but the documentation is from 2013 so might not be up-to-date.

Looking at [ros_comm](https://github.com/ros/ros_comm), it seems the `message_definition` is expected to be included in both publisher and subscriber (look in this [test for example](https://github.com/ros/ros_comm/blob/f5fa3a168760d62e9693f10dcb9adfffc6132d22/test/test_rospy/test/unit/test_rospy_tcpros_pubsub.py#L127)) and my manual tests seem to confirm it.

The reason this is an issue for me is that [Foxglove Studio](https://github.com/foxglove/studio) uses the `message_definition` to [determine how to parse an incoming message](https://github.com/foxglove/ros1/blob/main/src/TcpConnection.ts#L228). Since this is missing all messages become "empty" as well and an error is thrown. It works as expected for nodes using the official libraries which makes me think it is within spec to include it.

What do you think? Would it be feasible to add the `message_definition` to the header field? This is needed for me to be able to use node-to-node communication without using the ROS bridge in Foxglove Studio.

aler9, 2021-11-15:
This can be a good enhancement.

Edholm, 2021-11-15:
I agree. It seems that for example `rosbag record` complains when a topic is announced without the message definition being in place as well. It works, but not with a full feature set.

aler9, 2021-11-16:
`message_definition` contains the message definition and also its dependencies:

```
Header header

geometry_msgs/Quaternion orientation
float64[9] orientation_covariance # Row major about x, y, z axes

geometry_msgs/Vector3 angular_velocity
float64[9] angular_velocity_covariance # Row major about x, y, z axes

geometry_msgs/Vector3 linear_acceleration
float64[9] linear_acceleration_covariance # Row major x, y z 

================================================================================
MSG: std_msgs/Header
# Standard metadata for higher-level stamped data types.
# This is generally used to communicate timestamped data 
# in a particular coordinate frame.
# 
# sequence ID: consecutively increasing ID 
uint32 seq
#Two-integer timestamp that is expressed as:
# * stamp.sec: seconds (stamp_secs) since epoch (in Python the variable is called 'secs')
# * stamp.nsec: nanoseconds since stamp_secs (in Python the variable is called 'nsecs')
# time-handling sugar is provided by the client library
time stamp
#Frame this data is associated with
string frame_id

================================================================================
MSG: geometry_msgs/Quaternion
# This represents an orientation in free space in quaternion form.

float64 x
float64 y
float64 z
float64 w

================================================================================
MSG: geometry_msgs/Vector3
# This represents a vector in free space. 
# It is only meant to represent a direction. Therefore, it does not
# make sense to apply a translation to it (e.g., when applying a 
# generic rigid transformation to a Vector3, tf2 will only apply the
# rotation). If you want your data to be translatable too, use the
# geometry_msgs/Point message instead.

float64 x
float64 y
float64 z
```

aler9, 2021-11-20:
This feature has been implemented. Please test the library against Foxglove Studio and any other software that requires `message_definition` and let me know if everything works.

Edholm, 2021-11-20:
Nice! I will try it out and let you know

Edholm, 2021-11-21:
Looks to be working well, thanks for implementing this :100:

closed 2021-11-21 by Edholm