bluenviron/goroslib #112: ServiceClient.Call() need concurrency support

arboat, 2022-07-25:
## Which operating system are you using?

OS

- [x] Linux
- [ ] Windows
- [ ] macOS

Architecture

- [x] amd64
- [ ] arm64
- [ ] arm7
- [ ] arm6

## Describe the issue

The client side gets  "EOF" errors  when call  ServiceClient.Call()  concurrently, meanwhile the server side print : 

```
[ERROR] service provider '/test/xxxx' is unable to accept client '127.0.0.1:xxx': a client with id '/xxx' is already connected to the provider
```
It seems that this go ros package is not support concurrency? Would you please look into this problem, and give me some help to let me successfully call the function concurrently. Thanks a lot.

aler9, 2022-08-05:
Thanks, now `ServiceClient.Call()` is thread-safe.

closed 2022-08-05 by aler9

sniperZj, 2022-09-21:
@aler9 I updated it and the latest version will still appear

arboat, 2022-10-18:
@aler9 
Hello, I see you add mutex.Lock in the Call() method, actually it runs  sequentially. I used waitgroup to setup 1000 goroutines to call the method, and  it cost a lot of time.   So is there any possible to let the ros communication itself run  concurrently? Look forwarding for your reply. Thank you.

aler9, 2022-10-29:
> is there any possible to let the ros communication itself run concurrently?

You can't use a single connection to write requests and read responses concurrently, it's a limit shared by both ROS services and HTTP. 

Nonetheless, you can obtain concurrency by using a different `ServiceClient` in each goroutine.