bluenviron/goroslib #113: Strange behavior when publishing from windows and subscribing from linux

TheFern2, 2022-07-26:
## Which operating system are you using?

OS

- [x ] Linux, ROS Noetic Ubuntu 20.04
- [x ] Windows 10
- [ ] macOS

Architecture

- [ x] amd64
- [ ] arm64
- [ ] arm7
- [ ] arm6

## Describe the issue

Using the publisher example, I am able to publish from windows 10. When I start the publishing node, I can do `rostopic list` on linux (ros master) and see test_topic on the list, however nothing happens when echoing the topic. If I run the subscriber example same on linux, no data comes across, but see errors below. If I stop the publisher and do rostopic list then the topic is cleared out from the list as expected. 

Here's the kicker, if I run the subscriber from windows as well the publisher data comes across just fine. How is it possible that the data is going from windows to linux (ROS Master) and back to windows, but linux can't see the data? This is really strange, any ideas, can someone else test this scenario?

> I changed the message type from the examples to just a String message.

Windows 10 publisher and subscriber:

[image]

Getting this error on linux subscriber, where the roscore master is running, but `rostopic echo /test_topic` doesn't show any errors just empty terminal:

```
[2022/07/26 12:53:56] [ERROR] subscriber '/test_topic' got an error: Post "http://192.168.1.1:63795/RPC2": dial tcp 192.168.1.1:63795: i/o timeout
[2022/07/26 12:54:31] [ERROR] subscriber '/test_topic' got an error: Post "http://192.168.1.1:63795/RPC2": dial tcp 192.168.1.1:63795: i/o timeout
[2022/07/26 12:55:06] [ERROR] subscriber '/test_topic' got an error: Post "http://192.168.1.1:63795/RPC2": dial tcp 192.168.1.1:63795: i/o timeout
```

aler9, 2022-08-05:
Hello, i think that the error explains itself: the Linux node is unable to connect to the Windows node, probably due to a firewall in between.

The advertising of a topic (`rostopic list`) happens through a node --> master TCP connection, and evidently your Windows node is able to open a connection to the Linux Master.

The subscription to a topic (`rostopic echo`) happens through a node (subscriber) --> node (publisher) TCP connection. Your Linux node (subscriber) is not able to open a connection to your Windows node (publisher).

Just disable the firewall.

closed 2022-08-05 by aler9

TheFern2, 2022-08-05:
Thanks, I had added firewall inbound/outbound rules for the master ip, but I will try on another machine to turn off the firewall can't do it on the one at work.