bluenviron/goroslib #116: use rospy serviceProxy cycle call service err

sniperZj, 2022-10-08:
## Which operating system are you using?

OS

- [x ] Linux
- [ ] Windows
- [ ] macOS

Architecture

- [ ] amd64
- [ x] arm64
- [ ] arm7
- [ ] arm6

## Describe the issue
[image]

code:
#!/usr/bin/env python

import rospy
import rail_mounted_robot_msgs.srv

def call_get_cabinet_ext_data(room, cabinet):
    rospy.wait_for_service('data/get_cabinet_ext_data')
    try:
        get_cabinet_ext_data = rospy.ServiceProxy('data/get_cabinet_ext_data', rail_mounted_robot_msgs.srv.GetCabinetExtData)
        resp = get_cabinet_ext_data(cabinet, room)
        return resp.cabinet_name
    except rospy.ServiceException as e:
        print('Service call failed: {}'.format(e))

if __name__ == "__main__":
    rospy.init_node('test_client', anonymous=True)
    while not rospy.is_shutdown():
        call_get_cabinet_ext_data('20200331', '')

sniperZj, 2022-10-08:
service golib is the latest version

sniperZj, 2022-10-17:
@aler9 help

aler9, 2022-10-29:
Hello, goroslib's ServiceProvider is now able to accept multiple connections from the same node. Anyway, a node should instantiate ServiceProxy / ServiceClient once and reuse it, that error message is caused by the fact that ServiceProxy / ServiceClient was instantiated twice.

closed 2022-10-29 by aler9

sniperZj, 2022-10-30:
@aler9 Thank you