bluenviron/goroslib #119: After 17 days of continuous operation, some topics are published with a message indicating that i/o timeout and cannot be published again

wzbooks, 2022-11-24:
## Which operating system are you using?

OS

- [x] Linux

Architecture

- [x] arm64

## Describe the issue

Description
goroslib version:v0.0.0-20220623231355-7a670441a069

Hello, I have a problem with my program, I need your help, thank you

After 17 days of continuous operation, some topics are published with a message indicating that i/o timeout and cannot be published again, Data accumulation but failure to publish results in high CPU and memory usage

`[2022/11/22 16:33:38] [ERROR] publisher '/hardware_status' is unable to write to subscriber '127.0.0.1:36040' (TCP): write tcp 127.0.0.1:39843->127.0.0.1:36040: i/o timeout
[2022/11/22 16:33:39] [ERROR] publisher '/hardware_status' is unable to write to subscriber '127.0.0.1:36040' (TCP): write tcp 127.0.0.1:39843->127.0.0.1:36040: i/o timeout
[2022/11/22 16:33:40] [ERROR] publisher '/hardware_status' is unable to write to subscriber '127.0.0.1:36040' (TCP): write tcp 127.0.0.1:39843->127.0.0.1:36040: i/o timeout
[2022/11/22 16:33:41] [ERROR] publisher '/hardware_status' is unable to write to subscriber '127.0.0.1:36040' (TCP): write tcp 127.0.0.1:39843->127.0.0.1:36040: i/o timeout
[2022/11/22 16:33:42] [ERROR] publisher '/hardware_status' is unable to write to subscriber '127.0.0.1:36040' (TCP): write tcp 127.0.0.1:39843->127.0.0.1:36040: i/o timeout
[2022/11/22 16:33:43] [ERROR] publisher '/hardware_status' is unable to write to subscriber '127.0.0.1:36040' (TCP): write tcp 127.0.0.1:39843->127.0.0.1:36040: i/o timeout
[2022/11/22 16:33:44] [ERROR] publisher '/hardware_status' is unable to write to subscriber '127.0.0.1:36040' (TCP): write tcp 127.0.0.1:39843->127.0.0.1:36040: i/o timeout
[2022/11/22 16:33:45] [ERROR] publisher '/hardware_status' is unable to write to subscriber '127.0.0.1:36040' (TCP): write tcp 127.0.0.1:39843->127.0.0.1:36040: i/o timeout
[2022/11/22 16:34:03] [ERROR] publisher '/rosout' is unable to write to subscriber '127.0.0.1:34718' (TCP): write tcp 127.0.0.1:39843->127.0.0.1:34718: i/o timeout
[2022/11/22 16:34:27] [ERROR] publisher '/hardware_status' is unable to write to subscriber '127.0.0.1:36040' (TCP): write tcp 127.0.0.1:39843->127.0.0.1:36040: i/o timeout
[2022/11/23 06:37:35] [ERROR] publisher '/alarm' is unable to write to subscriber '127.0.0.1:35988' (TCP): write tcp 127.0.0.1:39843->127.0.0.1:35988: i/o timeout`

aler9, 2023-02-06:
Hello, subscribers are now disconnected after a write error, therefore this issue should be fixed.

closed 2023-02-06 by aler9

wzbooks, 2023-02-19:
Thank you very much

PR #129 (2023-02-06): publisher: disconnect subscriber after a write error
Fixes #119