bluenviron/goroslib #134: Unable to subscribe to sensor msgs or custom msgs, partial parse error

TheAdamP, 2023-02-24:
Hello,
When subscribing to topics containing specifically sensor_msgs/Imu or my own custom messages I get the same error.
`[2023/02/24 10:08:46] [ERROR] subscriber '/imu/data_raw' got an error: message was partially parsed, 1 bytes are unread`
`[2023/02/24 10:08:46] [ERROR] subscriber '/mcu/Status' got an error: message was partially parsed, 1 bytes are unread`
I have tried many fixes for this but nothing seems to be able to fix it including adding in queue sizes, making an empty callback function and changing the msg files. The error occurs in the line `Callback: func (msg *sensor_msgs.Imu) {`. Attached below is the code for the node and the header for the callback. The topics definitely have the right msg type as shown by rostopic info:
```
rostopic info imu/data_raw

Type: sensor_msgs/Imu
```
I am importing the sensor_msgs directly from the goroslib packages as well so everything should be fine there. All other msg types work as expected including std, geometry and nav msgs. As said before this also happens with a custom message made so there must be some correlation between them but I cannot understand how to go about fixing it. Any help would be appreciated.

Subscriber:
```
imuDataRaw, err := goroslib.NewSubscriber(goroslib.SubscriberConf{

		Node:     n,
		Topic:    "imu/data_raw",
		Callback: func (msg *sensor_msgs.Imu) {
			onSensorMessage(msg, "imu/data_raw")
		},
	})
	if err != nil {
		panic(err)
	}
	defer imuDataRaw.Close()

```
Callback:
`func onSensorMessage(msg *sensor_msgs.Imu, topic string) {
	...
	}
}`

wzbooks, 2023-03-01:
Hello, is the data sent using rosserial_server? I found that rosserial_server sends 1 byte more data, you may need to check the sent data

TheAdamP, 2023-03-01:
Thank you for the help, I believe this was the issue. I have since republished the data from a standard publisher in another C++ node and this now reads perfectly into the go code.

closed 2023-03-01 by TheAdamP