bluenviron/goroslib #175: package-import blew up

ppbrown, 2023-08-12:
## Which operating system are you using?

OS

- [x ] Linux
- [ ] Windows
- [ ] macOS

Architecture

- [ x] amd64
- [ ] arm64
- [ ] arm7
- [ ] arm6

## Describe the issue

Tried to use the latest package-import on one of our python-intended src directories for messages. It blew up.

Error output below.

Some background: It turns out the directory I passed in, is a top level directory, that holds a bunch of msg related definitions under it.
So there are subdirectories like

src/common_msgs/robot_supervisor_msgs/
src/common_msgs/ui_msgs

and those have the expected "msg" dir under them.

````
root@rostomqtt:~/workspace/go# package-import  /root/workspace/src/common_msgs
panic: runtime error: index out of range [5] with length 5

goroutine 1 [running]:
github.com/bluenviron/goroslib/v2/pkg/conversion.snakeToCamel({0xc0002eed27?, 0xc000302400?})
        /root/go/pkg/mod/github.com/bluenviron/goroslib/v2@v2.1.1/pkg/conversion/message_definition.go:69 +0x1cc
github.com/bluenviron/goroslib/v2/pkg/conversion.parseField({0xc000191665, 0xf}, 0xc0002fd9d0, {0xc0002eed20, 0x6}, {0xc0002eed27, 0x5})
        /root/go/pkg/mod/github.com/bluenviron/goroslib/v2@v2.1.1/pkg/conversion/message_definition.go:103 +0x7e
github.com/bluenviron/goroslib/v2/pkg/conversion.parseMessageDefinition({0xc000191665, 0xf}, {0xc000250df9, 0x9}, {0xc0001c2900, 0x228})
        /root/go/pkg/mod/github.com/bluenviron/goroslib/v2@v2.1.1/pkg/conversion/message_definition.go:220 +0x3ae
github.com/bluenviron/goroslib/v2/pkg/conversion.ImportMessage({0xc000250dc0, 0x46}, {0xc000191665, 0xf}, {0xc000191665, 0xf}, {0x606598, 0xc00009f5b8})
        /root/go/pkg/mod/github.com/bluenviron/goroslib/v2@v2.1.1/pkg/conversion/message.go:34 +0x105
github.com/bluenviron/goroslib/v2/pkg/conversion.ImportPackage.func1({0xc000250dc0, 0x46}, {0x6070d0, 0xc0002f0270}, {0x0?, 0x0?})
        /root/go/pkg/mod/github.com/bluenviron/goroslib/v2@v2.1.1/pkg/conversion/package.go:53 +0x525
path/filepath.walk({0xc000250dc0, 0x46}, {0x6070d0, 0xc0002f0270}, 0xc0000e1cc0)
        /usr/local/go/src/path/filepath/path.go:484 +0xff
path/filepath.walk({0xc0001ad640, 0x38}, {0x6070d0, 0xc0001af6c0}, 0xc0000e1cc0)
        /usr/local/go/src/path/filepath/path.go:508 +0x24f
path/filepath.walk({0xc000191640, 0x34}, {0x6070d0, 0xc0001af450}, 0xc0000e1cc0)
        /usr/local/go/src/path/filepath/path.go:508 +0x24f
path/filepath.Walk({0xc000191640, 0x34}, 0xc0000e1cc0)
        /usr/local/go/src/path/filepath/path.go:579 +0x66
github.com/bluenviron/goroslib/v2/pkg/conversion.ImportPackage({0xc000191665, 0xf}, {0xc000191640, 0x34}, {0xc000191665, 0xf})
        /root/go/pkg/mod/github.com/bluenviron/goroslib/v2@v2.1.1/pkg/conversion/package.go:40 +0x1c5
github.com/bluenviron/goroslib/v2/pkg/conversion.ImportPackageRecursive({0xc0000be180, 0x24})
        /root/go/pkg/mod/github.com/bluenviron/goroslib/v2@v2.1.1/pkg/conversion/package.go:113 +0x1b0
main.run({0xc0000a4010, 0x1, 0x1})
        /root/go/pkg/mod/github.com/bluenviron/goroslib/v2@v2.1.1/cmd/package-import/main.go:30 +0x112
main.main()
        /root/go/pkg/mod/github.com/bluenviron/goroslib/v2@v2.1.1/cmd/package-import/main.go:34 +0x54
````

aler9, 2023-08-12:
This was caused by the fact that one of your messages ends with an underscore. This is fixed and i released a new version (v2.1.2)

closed 2023-08-12 by aler9

ppbrown, 2023-08-12:
Thank you for the fast code update!
Just one more question (I'm relatively new to golang)

Your example doesnt say how to actually USE the generated msg package?

ie: I can now successfully call

package-import src/robot_msg_type

and that will create a new subdirectory "robot_msg_type", which is its own package.
But... I apaprently cant use either of

import "robot_msg_type"   

import "./robot_msg_type"

So, how do we use it please?

ppbrown, 2023-08-12:
Ah, I figured it out.
Have to use

import "main/robot_msg_type"

Could be nice if you added that to the top level README info on using package-import.

Thanks again for writing this, and the  quick update!

PR #176 (2023-08-12): fix crash when converting messages with a trailing underscore (#175)
Fixes #175