bluenviron/goroslib #195: Issue with IPv6 Address Parsing in Node Configuration

hesham1salem, 2024-02-05:
## Which operating system are you using?

OS

- [ x] Linux

Architecture

- [ x] amd64

## Describe the issue
I encountered an error when configuring a node in goroslib using an IPv6 address. Specifically, the application panics with the message when using the IPv6 address fc94:47d7:fc41:5099:e0cb:7560:2196:0647.
```bash
 "panic: listen tcp: address [IPv6 address]: too many colons in address" 
```
 Interestingly, altering afc94 with  4501  in the address  resolves the issue, suggesting a problem with parsing or handling the specific address.

```golang
package main

import (
	"github.com/bluenviron/goroslib/v2"
)

func main() {

	n, err := goroslib.NewNode(goroslib.NodeConf{
		Name: "goroslib_sub",
		//	MasterAddress: "[4501:47d7:fc41:5099:e0cb:7560:2196:0647]:11311",
		MasterAddress: "[fc94:47d7:fc41:5099:e0cb:7560:2196:0647]:11311",
	})
	if err != nil {
		panic(err)
	}
	defer n.Close()

}
```
the source of example 
https://github.com/bluenviron/goroslib/blob/ca98b523b42562e7c4abafb8a4d2615c296947b9/examples/subscriber-ipv6/main.go

aler9, 2024-04-21:
Thanks for reporting the bug, this is fixed by #203

closed 2024-04-21 by aler9

hesham1salem, 2024-05-07:
thank you for fixing the bug 
I wish if you can update the release so we can get  the package direct with go get command instead of clone the repo
 inside my project
 @aler9

aler9, 2024-05-07:
thanks for reminding me, i released the new version.

PR #203 (2024-04-21): fix listening on IPv6 addresses (#195)
Fixes #195